colin-campbell/firestudio-catalina #2: Catalina 10.15.6

burkebot, 2020-10-25:
Just wondering if there's another update that needs to be done on this. I'm currently on Mac OS Catalina v. 10.15.6 and my Firestudio project still won't connect. Is it because of my OS version? Is there an update coming? Thanks!

colin-campbell, 2020-11-29:
I shall check this on 10.15.7 as soon as I can.

johnhe4, 2020-12-02:
I just tested on 10.15.7, everything worked with my Presonus FP10. I ran the script, entered my password when prompted, allowed the exception in security center when prompted, waited about 2 minutes, then rebooted after everything completed successfully.

My FP10 was blue-light and I was smiles 🥇

colin-campbell, 2020-12-02:
Ah great. Thanks for testing, my time is limited. 
I recommend most people to run the DMG installer rather than the bare script, but I don't see any reason this shouldn't continue to work until Apple drop Intel or Firewire support altogether.

If it comes to that, I have kernel driver kernel signing privileges (these are very difficult to get from Apple nowadays). I will just write and release my own driver. I like my firewire gear and I hate forced obsolescence. with a passion.  :)

colin-campbell, 2020-12-03:
@burkebot Closing this issue since we have confirmed working on 10.15.7.
Recommend you uninstall everything with the Presonus uninstaller as detailed in the README, then reinstall using the DMG image installer. 
Pay special attention to allowing the firewire driver to run in Security Preferences.

closed 2020-12-03 by colin-campbell

burkebot, 2020-12-09:
Thanks for looking into this for me! A friend lent me their
Presonus Firepod, which worked no problem with your workaround. But my
Firstudio (which is newer than the Firepod), for some reason won't connect.
Anyway, I'll make due with this. Thanks again!