colin-campbell/firestudio-catalina #3: Change 11.1 -> 11.2

SaKiEQ, 2021-02-16:
Change 11.1 to 11.2 in main script to allow installation on latest version of Big Sur.

technicalboy-cmd, 2021-02-21:
Yes please, I am not able to run this on Big Sur 11.2.1

vicomte, 2021-03-02:
I was able to get my firestudio mobile to connect with the universal control and firestudio mixer eventually to work for 11.2.2 through a series of installing via command line and a few reboots. It seemed to take a bit for it to show up in my midi panel for audio inputs though.  [http://images.ravenhurst.com/fire.png](http://images.ravenhurst.com/fire.png)

SaKiEQ, 2021-03-03:
I just went into the script and changed the version numbers, which then allowed it to run and install on the latest OS build. 

Having said that, while it installs and runs, it seems to affect the system, the installed kernel extensions are no longer stable enough for a production machine sadly.

mgaspar1301, 2021-03-09:
> Change 11.1 to 11.2 in main script to allow installation on latest version of Big Sur.

I try with no success. How are the process?

mgaspar1301, 2021-03-11:
> Change 11.1 to 11.2 in main script to allow installation on latest version of Big Sur.

I try to change 11.1 to 11.2 with Script Editor without sucess. Can you help me?
I want help to install FireStudio 2626 on Big Sur. I pay for someone do it for me. 

Hugs.

SaKiEQ, 2021-03-11:
Change this:

if(!osver.startsWith('10.15') && !osver.startsWith('11.1'))

to -> 

if(!osver.startsWith('10.15') && !osver.startsWith('11.2'))

or which ever OS version you are running.... 

But I would suggest, if you don't know how to do this, then it might be better to stay away from it anyway. Catalina does not like the the old kexts very much and the system tends to become rather unstable.... so if you don't know what you are doing and how to revert/fix things when they break, then it would be safer for you not install this legacy driver.

mgaspar1301, 2021-03-11:
Thanks
I tried to change ’11.1’ to ’11.2’ without sucess. Same message.

Bad news of not being able to update the old kext.Here in Brazil it is very expensive to buy a new sound card of the same level as the FireStudio 26x26.For now I have a good installation of Mojave running satisfactorily.
A hug - Magnos

SaKiEQ, 2021-03-11:
not sure how to help you, I just run it again on 11.2 and it works on my
system without error.

What error message are you getting  ?

PS: I am running the script directly from the script editor.

mgaspar1301, 2021-03-11:
I wish you could enable the installation of FireControl on BigSur 11.2.3
Can you help me?

Here is the link to download the installation 
https://pae-web.presonusmusic.com/downloads/products/dmg/presonus_firecontrol_2626.dmg <https://pae-web.presonusmusic.com/downloads/products/dmg/presonus_firecontrol_2626.dmg>

A hug my friend.

colin-campbell, 2021-03-24:
> Change this:
>
> if(!osver.startsWith('10.15') && !osver.startsWith('11.1'))
>
> to ->
>
> if(!osver.startsWith('10.15') && !osver.startsWith('11.2'))
>
> or which ever OS version you are running....
>
> But I [...]

To be very clear. The firewire driver that is installed as part of this is the _official_ presonus supported driver from their own Catalina DMG. It is _not a legacy_ driver. You may download the new Presonus control panel and install the FW driver from that DMG yourself. However you will only have coreaudio support and no Universal Control 

What _is_ legacy here is the Universal Control package. And by legacy I mean unsupported. There is nothing incompatible with Catalina in it though, just the installer which fails with the packaged firewire driver (it tries to install the fw driver to a non-writable legacy directory and fails.

What this package does is extract 1) The current, supported firewire driver from presonus and 2) Extracts Universal control from the legacy package and installs that.

colin-campbell, 2021-03-24:
> I wish you could enable the installation of FireControl on BigSur 11.2.3 Can you help me? Here is the link to download the > installation https://pae-web.presonusmusic.com/downloads/products/dmg/presonus_firecontrol_2626.dmg <https://pae-> web.presonusmusic.com/downloads/products/dmg/presonus_firecontrol_2626.dmg> A hug my friend.

I shall look at this, given some time, but I will be working "blind" without the hardware. Unless this app is 64-bit, you will be definitely out of luck though :-/

colin-campbell, 2021-03-24:
To all. I have limited time and  only limited access to a Big Sur machine (I will not install it on my own mac until Apple make it less shit ) But I will try to make sure it works on the latest Intel versions as soon as I can

colin-campbell, 2021-03-25:
@mgaspar1301 Does the Firecontrol App have specific features for the 26x26 that Universal control doesn't have?

nighttraindj, 2021-03-25:
I have not tried this yet for my "Firestudio Project".  First, thank you Colin.  We all recognize that is in order.  I am grateful to just have a last shot at this.  Your talent for this is an asset to us.   

Concerns over the last few posts have me wondering two (likely insignificant) details:  

1.) Is this functioning if my intel iMac was never exposed to the old drivers that worked before the OS transition to Catalina?  (Was using Win. 7, but transitioned to this machine after that fell apart)?  I had read that having the Mac previously exposed to the drivers (before the update) made all the difference.  (Prior exposure=success).    

2.). Since this is a driver, Is it required that I begin with the Presonus "Firestudio Project" disconnected from the iMac.  Run the DMG (or copy/paste the code with update to the current .x version of Big Sur), then reboot the iMac, Finally, reconnect Presonus for the drivers to receive the new driver upload?  I would probably just do the install without disconnecting, but as it is uninstalling the old stuff for a clean slate are the extra steps necessary?  Too expensive to screw this up without asking.  

Plan B is to buy a brand new 1 in 1 out interface that will be far inferior to this pristine interface we are all sweating about and then move to spreading bad press about Apple everywhere I go if this fail.     

Note:  Obviously anyone reading this:  Turn off automatic updates download/install forever (as if you needed a reminder at this point, but this overhaul has kept me pretty busy lately and stuff that seems obvious can be overlooked.)  

Colin, Thanks again.

mgaspar1301, 2021-03-27:
> @mgaspar1301 Does the Firecontrol App have specific features for the 26x26 that Universal control doesn't have?

My hardware is the PreSonus FireStudio 2626. That´s the final driver for OsX Lion. With SiP disable, it works on Mojave with success. In the Big Sur, that´s not install. I want to enable it, but i don´t know how

nighttraindj, 2021-03-28:
Colin,

Sorry, my fault on not enough specifics.  I have Firestudio Project (not the 26x26).  I use the univ control app 1.76.  (Don’t know what the “Firecontrol” app looks like or even what it is.  Maybe I missed something about 10 yrs ago that would have added functionality, but I was formerly able to record MIDI and audio without it though MIDI controller setup and proper/reliable function has never been something that was working well.  Sorry if not more help and I’m not tech savvy enough to have good insight, but one think I am sure I have read was that the “hold control and click install” worked in Mojave (“hardened runtime support”? was supposed to go away with Catalina and/or Big Sur).  I think figuring out how to over-ride that is the equivalent of dis-enabling apple software security.  I wish you all the luck on that, but I think they designed it to prevent a work-around this time.  Maybe I read that from one of your posts or elsewhere, but shows you how little I actually know about how to fix this.  I’m sure the security can be or eventually will be cracked like everything else, but I assume it isn’t going to be an easy solution and Apple
will patch whatever holes they find ASAP.  The time between Mojave and now was designed to break through to encourage the software designers to issue updates and “legacy” hardware isn’t a priority for pretty obvious reasons....$$.  I keep finding myself chasing my tail on this incompatibility stuff and will probably leave things alone for awhile and decide what needs to be replaced or if I’m even going to attempt to get this rig up and running again.   I wish you luck and will answer any questions you might have, so feel free to inquire if you think it will help.  

Josh

colin-campbell, 2021-03-31:
> My hardware is the PreSonus FireStudio 2626. That´s the final driver for OsX Lion. With SiP disable, it works on Mojave with success. In the Big Sur, that´s not install. I want to enable it, but i [...]

That's not really what I asked. As far as I know, Firecontrol.app is very old and one should run Universal Control.app with the 2626. Correct me if I'm wrong. The FireWire driver is a different thing altogether from the App.

colin-campbell, 2021-03-31:
Firecontrol.app was a reply to mgaspar1301 
* Firestudio Project should work out of the box with this DMG, including on Big Sur, once I update the version numbers.
*  'hardened runtime" is only applicable to apps installed from the Apple App Store, not direct downloads.
* You should never have to disable SIP or anything to install this package. the Firewire Driver and UC are still signed by Presonus, and the installer DMG is signed by me and notarized by Apple.

nighttraindj, 2021-03-31:
Colin, 

Thanks again for helping me keep my studio running without overhauling basically every component that would become obselete in a sudden domino effect with the loss of this hardware function.  I’ll chime in again if I encounter any problems with Big Sur and using updated version # with the driver in console.  If all goes terribly wrong, there’s nothing left to lose for me on this rig, but sounds like this could save me a costly purchase if all goes well.  
Josh

colin-campbell, 2021-04-06:
Latest release should remove upper version check. Will close this ticket.

closed 2021-04-06 by colin-campbell

mgaspar1301, 2021-04-15:
> @mgaspar1301 Does the Firecontrol App have specific features for the 26x26 that Universal control doesn't have?

Is the latest version of the drivers supported by PreSonus for the FireStudio 2626 interface, which runs on macOS 10.7.5 through 10.14.6, disabling SIP. Universal Control have support at this v1.2.2715, Driver v3.5.5 (For FireStudio 2626)

mgaspar1301, 2021-04-15:
> That's not really what I asked. As far as I know, Firecontrol.app is very old and one should run Universal Control.app with the 2626. Correct me if I'm wrong. The FireWire driver is a different thing altogether from the App.

Great friend,

I think I understand better now. But isn't the latest version of Universal Control (for FireStudio 2626) the one in the picture?

(Appears on the PreSonus website) - A Hug
[image]