colin-campbell/firestudio-catalina #4: To uninstall, run the PreSonus uninstaller from the new dmg

studiotype, 2021-02-19:
Hi Colin, thanks for creating this! 

I'm trying to get my Firestudio Project working on Catalina 10.15.7, which I see should work from previous comments. 

I thought I'd remove an old drivers and try again, but re the readme: 
_'To uninstall, run the PreSonus uninstaller from the new dmg'_ 
I'm not seeing the PreSonus uninstaller in the dmg. Am I missing something?

colin-campbell, 2021-03-24:
Hi! Sorry for the late reply!

The uninstaller is on the new _Presonus_ dmg that is downloaded as part of this installation. Alternatively just download the latest installer DMG  for the PResonus website and run that. Their uninstaller script nukes both new and old versions of the driver!

closed 2021-04-06 by colin-campbell

studiotype, 2021-04-07:
And apologies for my slow reply!

I just tried this today – it's working, miraculous – thank you!

D