colin-campbell/firestudio-catalina #5: Congratulations

mgaspar1301, 2021-04-15:
Hello Collin-campbell

Now I was able to install version 0.3.0 of your installer. I couldn't test it because I don't have a PreSonus FireStudio Project card.

I would like to ask you if another project could be developed for the installation of FireStudio.app on the PreSonus FireStudio 2626 interface.

MacOS 10.7.5 Lion version app

Even Mojave, runs smoothly disabling SIP. But at BigSur, you would need a script like the one you did on this wonderful project.

I thank you very much and would pay you if you can help me.

A hug

[image]

mgaspar1301, 2021-04-15:
Can I use this script that you created to try to install my FireStudio 2626, even knowing that it would only reach an older version of UC?

I'll try it in the studio tomorrow and give you an answer. A big hug

mgaspar1301, 2021-04-15:
Hello partner,

I uninstalled everything from PreSonus and installed your script, however, on the Universal Control screen, FireStudio 2626 does not appear.

It appears on the "Sound" tab in System Preferences, however, there is no sound.

A hug

colin-campbell, 2021-04-20:
Hi,
I'm not familiar with Firestudio 2626 at all, are you saying it doesn't use the standard Universal Control, but rather a "Firestudio Control" app? Your screenshot looks like Universal Control.

mgaspar1301, 2021-04-22:
> Hi,
> I'm not familiar with Firestudio 2626 at all, are you saying it doesn't use the standard Universal Control, but rather a "Firestudio Control" app? Your screenshot looks like Universal Control.

Yes. In The Presonus site, in Firestudio 2626 download section are no Universal Control in The macOS 10.7.5 (Last oficcial support OS).

I want only a script that allow The install in 11.2.3 system. Can you help me? Hugs

mgaspar1301, 2021-04-23:
My dear friend Colin,

See The message in The realese notes of FireControl. Thanks

[image]

colin-campbell, 2021-05-03:
@mgaspar1301 I have added a test install for FireControl 2626 to the latest release (0.4.0) 

I can't check if it will work and unfortunately I don't think it will - the FireControl binaries and device are not signed by Presonus for running on later OSes and the SDK used to build the app is so old that I'm afraid Catalina+ won't run it.

I have added null signatures to FireControl.app and firestudiodevice.bundle and hope it works for you. If it doesn't I'm afraid there is nothing else I can do, sorry.

mgaspar1301, 2021-05-04:
``

> @mgaspar1301 I have added a test install for FireControl 2626 to the latest release (0.4.0)
>
> I can't check if it will work and unfortunately I don't think it will - the FireControl binaries and [...]

Hello Friend,

Thank you very much for the effort to make the FireStudio 2626 installation script.

Unfortunately, he hears some errors (as attached) and, as in the migration he had done, does not open FireControl (app) and there is no sound (although the interface driver is in the preferences)

Thank you very much for the effort.

A hug
[image]
[image]

colin-campbell, 2021-05-04:
@mgaspar1301 ah, I forgot you may not have the Apple developer tools installed to add the null signatures. Let me look at this to see what I can do. I really do not have much hope for this, FireControl.app fails to open the main window on my Mac (Catalina) and I think it is simply because it is too old.

mgaspar1301, 2021-05-04:
> @mgaspar1301 ah, I forgot you may not have the Apple developer tools installed to add the null signatures. Let me look at this to see what I can do. I really do not have much hope for this, [...]

How can I install Apple developer tools?

mgaspar1301, 2021-05-04:
> @mgaspar1301 ah, I forgot you may not have the Apple developer tools installed to add the null signatures. Let me look at this to see what I can do. I really do not have much hope for this, [...]

I believe that FireControl.app will only be shown if you have the hardware together.

Apart from the errors I showed you, you installed the firewire driver correctly in the System options. Only the sound does not come out.

If you can verify any more complements to what you have already done, I thank you immensely.

Hugs!

colin-campbell, 2021-05-05:
> How can I install Apple developer tools?

You can open a Terminal and type `xcode-select —install`  and just follow the instructions.
As described on this page. [As described on this page. ](https://macpaw.com/how-to/install-command-line-tools)

Then try the install process again.

mgaspar1301, 2021-05-05:
> @mgaspar1301 I have added a test install for FireControl 2626 to the latest release (0.4.0)
>
> I can't check if it will work and unfortunately I don't think it will - the FireControl binaries and [...]

Hello Master,

Attached are two screenshots where I use FireStudio 2626 on macOS 10.14.6 with excellent stability.

You may have some relevant information about the drivers that can help you in the development

Hug
[image]
[image]

colin-campbell, 2021-05-13:
@mgaspar1301 Have you tried to install the apple developer tools, as a I suggested and then reinstall? If it doesn't work, there is nothing I can do. Unfortunately I cannot legally resign and distribute Presonus code.

closed 2021-05-18 by colin-campbell