colin-campbell/firestudio-catalina #6: Apple Silicon Macs

aspearman, 2021-05-15:
Yes I know, you were probably dreading someone asking this. I'm not asking to support M1 Macs (yet?) 😁

I don't see a reason written down in this repo, and I'm just curious: why exactly will this never work on an Apple Silicon Mac? I presume the install location would be the same, but the application would run under Rosetta 2, right? 

I see PreSonus has released UC v3.4.2, which [adds support for M1 Macs](https://support.presonus.com/hc/en-us/articles/360051407452-Apple-Silicon-M1-Computer-Compatibility) (using Rosetta 2) to "all of [their] other USB audio and controller products" (though of course FireStudio isn't listed). I see from the [changelog](https://github.com/colin-campbell/firestudio-catalina/blob/master/CHANGELOG.TXT) you've bumped UC to v3.4.2. Perhaps v3.4.2 doesn't touch any of the software the FireStudio would need updated? In other words, PreSonus would need to fix their code, and we don't have access to the source code so we can't do it ourselves?

(Feel free to convert this to a GitHub Discussion.)

colin-campbell, 2021-05-16:
Hi. I download 3.4.2 to get the StudioLive Firewire driver which installs in the correct location for Catalina+. It is actually identical to the driver supplied with the old UC, with the exception that the installer installs it to /Library/Extensions as opposed to /System/Library/Extensions, which is now read-only.

The reason it will never work on M1 Macs is that this driver is Intel-only and Presonus will **not** develop an M1 version. From their release notes:

> StudioLive AI-Series mixers and rack mixers will not work with Apple Silicon Macs. There are no plans to develop a driver to enable 1394 to connect to these machines. We will continue to support Intel-based Macs.  

Because the FireStudio Interfaces use the DICE II chipset, they are not class-compliant Firewire devices and require a 1394 driver to operate. This will never appear on M1, unfortunately.

The absolute best one can hope for in this scenario is that someone develops a CoreAudio Firewire driver that supports a generic DICE II chipset by reverse engineering the old driver, and/or looking at the DICE driver for linux. Then you will get CoreAudio support, but no Universal Control. All this is assuming that Apple doesn't drop Firewire audio support altogether in the near future. 
I have looked all over the internet for the TC Electronics DICE II SDK toolkit with no luck

aspearman, 2021-05-17:
Interesting. I don't know a lot about driver software, but I think I'm following what you're saying: essentially because the DICE II chipset isn't class-compliant with IEEE 1394 (aka FireWire 400) it needs a driver to be installed, and PreSonus doesn't plan to update their now-ancient driver software to work on Apple Silicon (despite updating some other products' software for Apple Silicon support in UC v3.4.2). 

But PreSonus isn't the only one that could help. Apple created a DriverKit framework in 2019, but despite the upset posts on the Developer Forums, it doesn't look like it'll support FireWire. Otherwise, the other solution would be developing a generic DICE II driver somehow to at least get CoreAudio support. 

Apple is phasing out Intel Macs (overall for the better IMO), so it's only a matter of time... with that context, they really should support FireWire in DriverKit. Though if DICE II isn't class-compliant this wouldn't solve all the issues would it?

Such a waste of perfectly functioning hardware due to lack of software support. Perhaps I should be trying to future-proof better by buying class-complaint interfaces that don't completely sink with the ship when the company declares it end-of-life i.e. doesn't rely on installing some software from the company. 

Thanks for the info though; interesting to hear what the actual problem is.

closed 2021-05-17 by aspearman