colin-campbell/firestudio-catalina #7: Firestudio Project

torava, 2021-05-17:
Hi, after upgrading from Catalina to Big Sur 11.2.3 I haven't gotten my Firestudio Project working. I managed to install new version of Universal Control but when I connect Firestudio Project, it's light will remain red and it won't appear on my computer.

colin-campbell, 2021-05-18:
You need to be more specific. Where did you install a "new version of universal control" from?

torava, 2021-05-19:
I unistalled old Universal Control which I had for Catalina and then tried both your releases v0.3.0 and v0.4.0 with no luck.

vicomte, 2021-05-19:
I had it running in 11.2 but today in 11.3.1 I only get a red indicator on my firestudio mobile (usually got blue or red/blue alternating). kernel ext shows loaded and system report shows an unknown firewire device. Reinstalled driver from the sh script just to exhaust possibilities.  I will keep tinkering with it.

AnAvidEditor, 2021-05-21:
Hi! I am having the same issue. Have a Presonus Firestudio Project On Big Sur 11.2.3 and getting no response after installing. The light stays red and macos doesnt recognize it. It worked on earlier versions. Thanks!

colin-campbell, 2021-05-22:
Alright. Thanks. I will test 11.2.3 as soon as I can! You should always use the oldest supported Apple system possible. there is absolutely no need to upgrade to the latest and greatest OS Release in a DAW, unless you have software that won't run on the older system.  Upgrading immediately on a working system is asking for trouble.

AnAvidEditor, 2021-05-22:
Thank you! Yes I definitely agree. It’s actually my brother’s MacBook and
I’m helping him out. I’ve told him not to upgrade...

colin-campbell, 2021-06-03:
If the light is red or flashing then the Firewire driver isn't loaded. Check your System Preferences -> security settings in case you have to manually give permission for the driver to run. I have UC and the driver running  on 11.4 without problems. But not a Firestudio project.

https://user-images.githubusercontent.com/1169963/120710956-18467380-c4bf-11eb-80f0-8900b135c8ec.mp4

AnAvidEditor, 2021-06-03:
Ok awesome. Thank you for checking into it. I’ll take a look.

closed 2021-07-08 by colin-campbell

henry2018k, 2022-06-11:
Ho anche io una presonus firestudio project con big sur,ma ogni tanto quando carico logic si spegne audio in uscita,può essere il driver?