colin-campbell/firestudio-catalina #8: Strange issue

TreyM, 2021-05-24:
I'm on 11.4 (developer beta needed for my AMD GPU, 6800 XT to be supported) and my Firestudio Tube works as expected until I open Logic into a new project. Then all outputs die, but inputs still work. This is affected system wide until reboot. I'm not sure this is something that can be fixed though...

colin-campbell, 2021-06-03:
It's working on the 11.4 release version for me without problems, although I don't use Logic. If it was a driver crash you would know all about it and your machine would reboot, but maybe you can see something in Console.app if Logic is having problems with the audio HAL. Might even be a bug in Logic. 

https://user-images.githubusercontent.com/1169963/120710623-a4a46680-c4be-11eb-82a8-39f6bd5ea46a.mp4

TreyM, 2021-06-07:
I think I have narrowed down the issue. Any time I change sample rate, be it system wide using the Audio MIDI Setup app in the macOS Utilities folder or within Logic itself, the outputs stop working for me. Can you reproduce this on your system? (Full disclosure, I'm on a hackintosh system so this may have something to do with my issues, dunno)

colin-campbell, 2021-06-19:
I can't reproduce with the setup I have, but try starting up Universal control and changing the sample rate of the interface there before attempting anything else. Not sure it'll help with your problem, but worth a try.

closed 2021-07-08 by colin-campbell