colin-campbell/firestudio-catalina #10: 'fsprojectdevice.dll' Library missing.

Vinseph, 2021-06-30:
Hi,

Thanks, in advance, for all the great work here!
Tried your install for my firestudio 2626 on MacOS 11.4 (Big Sur). Installed both UC and FireControl for my 2626. Either of them always produces red flashing light on my hardware. Never a solid blue. Tried it with my csrutil both enabled and disabled in safe mode. All the same results. When I open UC, it states 'Can not find required Library "fsprojectdevice.dll". Please re-install'

Any tips of what to specficially check to make sure the driver is working? I never have any issues with your install. It always completes. 

Thanks!

Best,
Vincent

Vinseph, 2021-06-30:
Just realized... dll files are for Windows machines, no? How can I have a dll issues on my MacOS? 🤔

colin-campbell, 2021-07-03:
I'm afraid the 2626 will have to remain unsupported. No-one has been able to get it to work, unfortunately.

Vinseph, 2021-07-07:
Thank you for the update. Much appreciated.

closed 2021-07-08 by colin-campbell