colin-campbell/firestudio-catalina #11: macOS Big Sur V11.4 Issue

NickC23, 2021-08-04:
Hi! Unfortunately my Mac ran an auto-update to the latest OS (11.4) which has seemed to break the patch/hack of the Presonus installer. Universal Control does still open without fail but does not recognize my device (FireStudio Mobile).  My interface is still getting power from the thunderbolt so not an issue on the connectivity front.  

I am running the Mid 2014 15" MacBook Pro with Retina.  Thank you so much for the hack and for taking a look at this.

colin-campbell, 2021-08-06:
@NickC23 if you have uninstalled and reinstalled, recommended you check your "System Preferences->Privacy & Security" page for information that looks like this.
[image]

After upgrading (to 11.5) I had to re-authorize certain  extensions:

[image]

Let me know how you get on.

colin-campbell, 2023-01-02:
Closing as confirmed working up to Ventura 13.1

closed 2023-01-02 by colin-campbell