colin-campbell/firestudio-catalina #12: No audio on Big Sur

kogitune55, 2021-09-20:
Hi. 
Thankfully my FireStudio Mobile works well on Catalina.

On Big Sur, it gets no sound after the sleep mode.
The lamps lights up correctly in blue but audio signals seems lost inside Mac.

Does someone have the same  trouble or know how to solve it?

Thank you.

colin-campbell, 2021-09-22:
Thank you for this. I will test.
I actually haven't tested sleep/wake, since I never let my audio computer sleep. Using power-management on a DAW is something I gave up on years ago :)

kogitune55, 2021-09-23:
Oh, GLAD for your testing.

In Japan where I live, FireStudio Mobile is hardly used these days, so your Catalina driver saved my music life.

Thank you so much.

colin-campbell, 2023-01-02:
@kogitune55 this is confirmed working up to Ventura 13.1. And I cannot reproduce the sleep issue, so I shall close this.

closed 2023-01-02 by colin-campbell