colin-campbell/firestudio-catalina #13: Getting it to work on Monterey 12.2.1

blisst, 2022-03-08:
This issue is informational for anyone who has problems on Monterey 12.2.1.

When I first upgraded to Monterey, connecting my Presonus Firestudio was freezing my system when it was plugged in and powered on.

## TLDR: The answer was to make sure all aggregate audio devices in Audio Midi Setup were *NOT* using the Firestudio before plugging it in.

If that doesn't work for you, it may have helped that I did the following to reset permissions on the Presonus `kext` files.

1. Uninstalled the Presonus Firewire drivers via Presonus's official DMG.
2. Restarted into Recovery Mode (hold Cmd + R on restart)
3. Go to Utilities > Terminal at the top.  
4. Run `kmutil trigger-panic-medic --volume-root /Volumes/YOUR_HD_DRIVE_NAME` which I found [from this guide](https://forum.rme-audio.de/viewtopic.php?id=33277)
5. Restart
6. Reinstall drivers using the latest release from this repo (0.4.0 at time of writing)
7. Restart
8. Go to Security & Privacy, click Details next to the "software not being allowed" part at the bottom. 
9. Make sure Presonus is checked (allowed) and hit OK
10. Restart

masoncj, 2022-12-30:
Can confirm working on Monterey 12.6.2 with Firestudio Project on MacBook Pro 2015.  Only needed to enable the extension in Security & Privacy > "Some system software requires your attention..." > Details > Presonus.

Thanks so much!

colin-campbell, 2023-01-02:
Closing this as confirmed working up to and including Ventura 13.1

closed 2023-01-02 by colin-campbell

fotodisegno, 2023-01-30:
I have a problem.
I downloaded the latest version 0.4.1. An error come up. 
The script couldn't fetch the driver on presonus. ERROR -270