colin-campbell/firestudio-catalina #14: Big Sur 11.7.1

PacBilly, 2022-11-29:
Hi. I'm running a Firestudio Project on Big Sur 11.7.1 on my Mac Pro 5,1 under OpenCore. The Firestudio works fine with your patch under Catalina on my MacBook, but not on the Mac Pro. I've configured OpenCore to disable SIP. I've uninstalled and reinstalled using your instructions. The System Profiler detects the interface and knows what kind of card it is, and a listing of third party kexts shows that the kext for the Firestudio is loaded, but neither the system or universal control acknowledge its existence. 

Any ideas what else I might try?

PacBilly, 2022-11-29:
It should also be noted that when I boot into Mojave, the interface works fine.

colin-campbell, 2023-01-02:
Hi, you should not need to disable SIP to install this. (The driver packages are signed)
I will check with my own MacBook  when I get the chance.

colin-campbell, 2023-01-02:
@PacBilly I have just checked with all versions up to Ventura 13.1, and it works.
Unfortunately I can't test on a OpenCored Chees-grater Mac, so I shall close this.

closed 2023-01-02 by colin-campbell