colin-campbell/firestudio-catalina #15: can't find variable: button

PacBilly, 2023-01-10:
Shortly after launching the newest version 0.4.1, the download progress bar is shown briefly, then the following error is shown under Big Sur 11.7.2. 
Error: ReferenceError: Can't find variable: button (-2700)

It presents the choice to edit the script or press OK, which closes the application.

micahjdailey, 2023-01-16:
Having the same issue. Any tips on out to fix this  would be amazing.

LLBB73, 2023-01-20:
[image]
The same here ! With OpenCore Legacy Patcher Monterey 12.6.2

PacBilly, 2023-01-26:
I've confirmed that the same issue persists regardless of whether you're using OCLP or not. Native installs throw the error as well.

fotodisegno, 2023-01-30:
I'm using Monterey 12.6.3

fotodisegno, 2023-01-30:
EDIT:
I downloaded the 0.4.0 version and everything was ok.
I'm using Monterey 12.6.3 and a Firestudio Project.
Thanks a lot.

colin-campbell, 2023-02-16:
Probably a brainfart from me removing the 2626 installer. There should be no driver difference using 0.4.0 for the moment, but I'll get a fix out for this.

boyz001, 2023-02-25:
Hi The Same For me on the newest open core ! was Going to buy you a Coffee ! Lol

boyz001, 2023-02-25:
Ive tried the 0.4.0 one and seemed to install ok button sound ! See attached Pic ! I must be doing something wrong ! Thanks so much for trying to sort This out. I Just spent £500 to have it fixed then find out it won't work on Monterey !

boyz001, 2023-02-25:
[image]

boyz001, 2023-02-27:
will this work on Monterey 12.6.3 ?

colin-campbell, 2023-03-03:
The "button issue" should be fixed in v0.4.3. I shall close this. If you have another problem, please open your own issue.

closed 2023-03-03 by colin-campbell

burkebot, 2023-03-03:
I'm still getting the same issue with v.0.4.3

fotodisegno, 2023-03-04:
Use the 0.4.0 version. Works fine

boyz001, 2023-03-04:
Thank you for your reply I’m trying with Monterey with FireWire I’m getting
input but no output would it be better if I got a thunderbolt card with two
adapters to convert ? It’s seeing the FireWire device but I have tried
everything and am at a loss ! Any help would be greatly appreciated! John

fotodisegno, 2023-03-04:
What kind of Mac are you using?.
I'm using a Mac Pro 6.1. It has already Thunderbolt 2.
First I was using several adapters: Thunderbolt 2 to Firewire 800 (Apple Original, don't waste money on cheaper ones) and then another adapter from FW 800 to 400. Otherwise there's a good cable on Amazon FW800 to FW400.
Now I'm using an OWC Dock Station Thunderbolt 2 that it has a FireWire 800 port. So I need only the FW 800 to FW400 cable.
Are you using the 0.4.0 version?. That version is PERFECT for Firestudio

boyz001, 2023-03-04:
Thanks for the reply.I’m using a Mac Pro 5.1 and trying to run Studiolive
16.0.2 desk with just the FireWire cable as I don’t have thunderbolt card !
I have a usb c / usb3 card and tried with the apple approved converters but
that doesn’t work so it seems I might have to get the thunderbolt card or
look into the dock station you mention.
 I’m running your 0.4.0 version

boyz001, 2023-03-04:
Thanks Colin-Campbell/Firestudio-Catalina

boyz001, 2023-03-04:
Am I correct in thinking the dock would be better ?

fotodisegno, 2023-03-06:
@boyz001, you're using Monterey on a Mac Pro 5,1?. Means somehow you hack the computer. 
Maybe, and I say maybe that's the problem. I had a 5,1 and the newest macOs you can use is Mojave. The only way to step up is hacking the system.
I don't know if the dock will solve the problem. You will need a thunderbolt card and the dock. With the money you will have to spend on this item is better to find a used Mac Pro 6,1.