colin-campbell/firestudio-catalina #18: Drivers Sometimes don't load after reboot

frank6311, 2023-05-22:
This may not even be an issue with your Fix.  May be a macOS issue but after a reboot the drivers don't load and so the Presonus UI seems to crash.  After second restart the drivers appear to be fine and everything works as expected.  Looked for a solution but could not find an answer as to why it happens.  As  I said it may be a macOS issue anyway.
thanks for the great work btw.

colin-campbell, 2023-05-27:
Which OS version are you using? If the sync light is flashing and not steady on the hardware, then the drivers have not loaded.

colin-campbell, 2023-07-08:
The only message I usually get from time to time, is "This driver may be incompatible with a future version of MacOS" after an install. Is there a specific message?

closed 2024-01-23 by colin-campbell