colin-campbell/firestudio-catalina #19: Driver Won't Load on Ventura 13.4

AdamPavelec, 2023-06-13:
Greetings and many thanks for developing this script! I cannot seem to get the driver to load on my 16" 2019 MBP I7. I have ran the uninstaller from the DMG (which was tricky, as it seems to disappear from ~/Desktop rather quickly), rebooted, installed from the DMG, rebooted, and power-cycled my interfaces (I have two linked together). I am using a brand new USB-C -> Thunderbolt dongle connected to a brand new Thunderbolt -> Firewire dongle, both from Apple. Universal Control reports driver version 0.0.0.0 5875. System Information > Hardware > Thunderbolt/USB4 reports no device connected on both busses. Any ideas?

colin-campbell, 2023-07-08:
The DMG shouldn't disappear! 
The only thing I can think of is the extension requires your permission to run.
https://support.antelopeaudio.com/en/support/solutions/articles/42000098875-how-to-enable-the-unified-driver-on-mac-os-ventura

colin-campbell, 2024-01-02:
@AdamPavelec I have tested now up to Sonoma and still working. If we can't find the root of your problem, I shall close this.