colin-campbell/firestudio-catalina #20: Firestudio Project Inout are Working but outputs are Not

Linuxvst, 2023-12-22:
Hi Collin i tried many times with different versions from v0.2.0 to 0.4.4
the inputs are always Working the driver Loads the Blue light is lit on the Interface

i also tried resetting the Driver by running 
kmutil trigger-panic-medic --volume- root /Volumes/Macintosh_HD as Specified as a workaround 

when i reboot i go to Security & Privacy and Allow the paefirestudio.kext  again re-power the interface the driver loads and the blue light stays On 
but still no Sound on the Output

i tried on 2 different mac Pros and on 3 operating System Monterey Ventura and Sonoma 
i was able to get it to work only Once on Sonoma with v0.4.0 i rebooted and lost it Again
if i go back to catalina with previous drivers  0.2.0 and it still works Perfectly 
but from Monterey and Up it is Not Working

i do not know what else i can try!
 i also tried to install the FireWire drivers,  launchdaemon ect.  Manually to there proper locations from UC v3.4.2-63992 (extracted with Pacifist)
 and the Universal Control App From v 1.7.6-5875 (extracted with Pacifist)

driver loads Blue light is on steady but i still have the same results all Inputs are Working but no Outputs
and not even on output 1-2 without the Mixer

i hope you can find a way to get it to Work 

Thank's in Advance

Michel

colin-campbell, 2023-12-29:
Hi @Linuxvst , if the blue light is steady, then the driver part is up and running ok. Can you see the outputs in Audio/MIDI setup?

Linuxvst, 2023-12-29:
Hi Collin yes the lights are steady and i see the outputs in audio midi Setup i also uninstalled any other audio Drivers and aggregate devices i tried on 4 Different machines  2 mac pro 3.1  and 2 mac pro 5.1 all Oses installed with the latest Open core Legacy Patcher  tried from Monterey up to Sonoma all the same no Audio Output from Core Audio
 i verified to see if the Proper universal Control was also installed  i have  Universal Control 1.6.5875
 i'm used to work around issues but as i told you if i feed an input with a signal the inputs are working and going thru the output 1-2 but if i use any software like Logic or even with audio midi Setup i get no sound from the outputs even i i play along with the universal Control i tried all other outputs to see if it was a bad routing and no signal is going thru from mac os CoreAudio

Linuxvst, 2023-12-29:
By the way just to tell You i also Have an old  Digdesign Digi 003 Firewire Interface  and i was able to create an installer  and 
 ake it work on all systems from Catalina  up to Sonoma

Linuxvst, 2023-12-29:
i also deleted the com.apple.audio.DeviceSettings and com.apple.audio.SystemSettings in Library/Preferences/Audio 
in case of corruption  i rebooted new .plist files but same Problem!

Linuxvst, 2023-12-29:
My Interface is Firestudio Project 8 Inputs 10 Outputs

colin-campbell, 2023-12-29:
Interesting. I can only test on my firestudio mobile and firestudio tube, but have no problems. 
If you can see the audio devices in audio midi setup, I can't think if what should cause that, unless there's some bizarre config in universal control or something. Try it making sure UC is set to bypass.

Also try changing the sample rate in the UC control panel. if that works, then I'm pretty sure the problem lies elsewhere, because it means the HAL plugin is talking to the driver correctly.

colin-campbell, 2023-12-29:
I have to add, I have not tried this at all on Hackintoshes or anything using OpenCore

Linuxvst, 2023-12-29:
it.s not a hackintosh they are mac Pro's  i also tried changing the sample rate the interface receives the signal and changes the sample rate on the interface and you said HAL plugin ? i never saw a HAL plugin for Presonus 
the only thing i think that is not doing the job is the fsprojectdevice.bundle i do not know were to check to see if it loads

P.S. i checked that the paefirestudio.kext it is loaded and hat the dependency are fullfilled

Linuxvst, 2023-12-29:
Here are Some Screen shots
[image]
[image]
[image]
[image]

Linuxvst, 2023-12-29:
i also tried with different Firewire Cables firewire 400 to firewire 400, firewire 800 to firewire 400
 P.S. tested the firestudio project with all these cables are it's always Working when i Boot in Catalina or even Older El Capitan

colin-campbell, 2023-12-29:
I have a trashcan Mac Pro from 2013 running Monterey. No problems with the Firestudio Tube. Not sure if OpenCore could affect it, but see here: 
https://forums.macrumors.com/threads/firewire-audio-issue-with-mac-pro-5-1-with-opencore.2381798/

Linuxvst, 2023-12-29:
Hi Collin i Checked the link above i tried the quirk (disableIOmapping) it;s not working!
and  it will never work because they say for real macs you need a 2012 or later Mac
you need a motherboard that supports VT-d 
so Catalina is the Highest for macs under 2012
Thank's for your Time  pass the word around so others don't lose time on trying!

colin-campbell, 2023-12-30:
Thanks, I'll add that to the README, and close this.

closed 2023-12-30 by colin-campbell

Linuxvst, 2023-12-30:
Hi Collin you made a Mistake In Writing the note 

You Wrote 

Note: The driver will seemingly only run properly on pre 2012 Macs up to Catalina

it Should Be post 2012  instead of pre 2012!

 

 

   Michel Bibeau

Linuxvst, 2023-12-30:
I made some other tests I have  one mac Pro 5,1 with an upgraded Xeon Processor that Supports VT-d Xeon X-5675 

Even using the Quirk disableIOmapping in the config.plist it is still not Working 

I think  It is really a mapping problem !

 

so the explanation your wrote is quite not correct 

it is not a hardware problem because the interface is working with the same drivers on the same machine up to Catalina 

it is just that apple Changed something else in the software part the way that the driver Communicates with the Interface 

so it is really a driver problem we would need a re-write of the driver itself 

an installer with corrected  installation locations for this is not working

 

P.S.

 I already made 2 Corrected installers for the Digidesign  “Digi 003”  and the Apogee “Rosetta 800” ( using old snow Leopard Drivers) 

and they are working properly on my mac pro 3.1 and 5,1 machines up to Sonoma 

 

too bad this does not work for Presonus with those old drivers!

Anyway Good Work for those who it is working For!

If you have any other idea let me Know!

 

 

 

   Michel Bibeau