colin-campbell/firestudio-catalina #21: How to clean uninstall the driver ?

Arthurdhaeyer, 2024-01-23:
Hi Colin, your driver is amazing, I used it for a year with both a Firestudio Mobile and Project : perfect !
But two weeks ago, overnight, I couldn't use my devices anymore, for no reason !
I tried may things, and my cards are ok (tested on another computer).

I deleted the firmware downloaded a year ago and installed the somewhat new version, it did not solved my problem.
Maybe I deleted it wrong ? How to clean uninstall the program ? I don't see any Uninstall button or file...
thank you very much Colin !

I also sent you an email with more details at colin.campbell@d********group.com

Arthurdhaeyer, 2024-01-23:
sorry I didn't mean to close the discussion, just clicked randomly on the button

colin-campbell, 2024-01-24:
@Arthurdhaeyer Your fw port should be ok.

You can check if the driver is loading by opening a terminal and issuing the command:
`kextstat | grep PaeFireStudio` 

You should see output similar to this (not identical) if the driver is loaded.
`  180    0 0xffffff7f962a4000 0x17ffc    0x17ffc    tc.tctechnologies.driver.PaeFireStudio (4.2.1 46437) 9D76C25B-A48C-34EF-9E49-B72392F3199C <179 178 7 6 3 1>`

Arthurdhaeyer, 2024-01-24:
Hi, thanks for your answer,
when I do `kextstat | grep PaeFireStudio` : nothing happens
when I do `kextstat` : I see 179 lines and no mention of PaeFireStudio...

strange ?

[image]

Music is my work so I am stuck in this weird situation, thank you very much for your attention Mr Campbell !

Arthurdhaeyer, 2024-01-31:
Hi Colin, do you have any ideas what I could do next ? I am stuck...
Thanks a lot for your help ! :)

Arthurdhaeyer, 2024-02-11:
Could you tell me how to properly uninstall the driver, maybe there is a command to type or smth to delete in Library, I don't know. I feel stuck, music is my work the problem is I can't afford a new macbook pro and it feels crazy to buy new soundcards as those are working perfectly fine... thanks again for your help, sorry to disturb you but you are the only person I know who can understand a word of this !
best regards !

Arthurdhaeyer, 2024-02-23:
WUUP ! I managed to solve my problem !
[image]
What i did was simple, I deleted the app simply by drag and drop to the bin, empty the bin, turn of the computer, do SMC and PRAM reset, turn on the mac, download and install your firmware, reboot the mac, then it was working like it did !!
I feel so relieved !

closed 2024-02-23 by Arthurdhaeyer