colin-campbell/firestudio-catalina #22: driver seems to load, but no sound and no output/peaks in Universal Control

russendisko, 2024-02-06:
Hi Mr. Campbell, 
I had to update my mac pro 5.1 from Mojave to Monterey. It was a clean install.

First I installed the driver and software with your tool. I got sound, all fine. 

After that, run the Migration of my backup system (Mojave running also the presonus firestudio mobile).

After restart i tried again, no sound, the Soundcard is recognized in audio midi setup, also blue light at the unit, but no sound and the universal control gets no peaks when I play music.
I deinstalled all via the Presonus dmg and did also the cleaning of the kext in recovery mode. new installation same problem.
kext is loading:
Executing: /usr/bin/kmutil showloaded
No variant specified, falling back to release
  155    0 0xffffff7f9a393000 0x18000    0x18000    tc.tctechnologies.driver.PaeFireStudio (4.2.1 46437) 9D76C25B-A48C-34EF-9E49-B72392F3199C <154 136 7 6 3 1>

maybe you can help? 
Thanks in advance

russendisko, 2024-02-07:
addional Info, when I play sound thru (line in) the card there is a output on the universal control panel

russendisko, 2024-02-10:
another update, it is possible to selected the presonus firestudio moblie in a daw and record sound also there is a visible output in the meters of the daw. I think the driver works but there ist an output conflict in mac os i don't find.

colin-campbell, 2024-05-03:
@russendisko I don't use UC myself, just connect the DAW straight to it. I suspect you might have imported som UC settings from your backup which are confusing UC. 
Try this:
* Make sure UC is closed completely
* In finder menu, select "Go -> Go to Folder..."
* in the dialog, type this exactly: ~/Library/Preferences/PreSonus
* Delete the folder "Universal Control (x64)" 
* Restart Universal Control and select your soundcard in the control panel.

Hope that helps!

russendisko, 2024-05-06:
Thanks Colin, but unfortunately no sucess

colin-campbell, 2024-05-16:
@russendisko Are you getting metering in Universal Control when it is open?

russendisko, 2024-05-16:
only incoming sound is displayed (also out when I run the sound directly thru the soundcard), but no metering in for internal sound for Out 1-6

colin-campbell, 2024-05-16:
That sound is being processed at all and is being displayed in UC means the driver is loaded properly. It pretty much has to be a configuration or routing problem at this stage.

russendisko, 2024-05-17:
i mean incoming audio signals from external sources, which i route from the audio in through the sound card (this is displayed in the universal control). processed sound from my daw, youtube, audio player etc. is not displayed

colin-campbell, 2024-08-19:
@russendisko Did you get this solved? There's not much I can do here, so will close this.

russendisko, 2024-08-20:
nope, i think it's a oclp problem, thank you for your efforts

colin-campbell, 2024-08-22:
Sorry I couldn't help!

closed 2024-08-22 by colin-campbell